mjs513/Teensy-4.x-Quad-Encoder-Library #2: Question: How well does this library handle bounce?

stevewells20, 2020-03-16:
I've used Paul Stoffregen's Encoder library for a long time, and it has always worked really well. I see that this library uses the hardware encoder channels on the Teensy 4.0, but I don't see how it handles bounce issue (like setting sensitivity for the pulse count), other than maybe changing the filterSamplePeriod. Do you have any information/experience with this?

mjs513, 2020-03-16:
I have used the same library for years as well and I agree it does work
really well.  As for your question on bounce.  Changing the
filterSamplePeriod and changing the filterCount are the only ways
internally that can address debounce.  The IMXRT1062 filterCount is a
minimum of 3 and maximum of 7 if you do use it.

If you do a search on the forum the subject has been brought up before, for
instance:
https://forum.pjrc.com/threads/59180-Encoder-library-with-cheap-encoders?highlight=encoder+debounce
If you do a search you for debounce circuit you will find
https://circuitdigest.com/electronic-circuits/what-is-switch-bouncing-and-how-to-prevent-it-using-debounce-circuit

laudiblellc, 2020-05-27:
I've just had some experience with this issue. I'm using one encoder channel to read a quadrature motor encoder, and the other channel receives high frequency step/dir commands. Without any filtering configured i was getting a lot of inaccuracy with my step/dir. The smallest filtering settings completely fixed the issue and things are running smooth as silk.

mjs513, 2020-05-27:
Its good to know that the option is working for you laudiblellc. With the cheap KY-040 Rotary Encoders I had to use the highest filter settings.  Going to close this out.

closed 2020-05-27 by mjs513