mjs513/Teensy-4.x-Quad-Encoder-Library #3: Question: Index Interrupt

josiah47, 2020-03-26:
Is it possible to link index trigger to interrupt routine?

I would like to digitalWrite when index is seen...

mjs513, 2020-05-27:
Would think it would work, never tried it.  Give it a try and let us know.

LMFLox, 2022-03-23:
Do you think is possible to take a snapshot of the position counter when index trigger is rising up instead of the revolution counter [in/de]crement?

mjs513, 2022-03-23:
I would think you can do it from within your sketch.  If you have a variable like oldIndex - curIndex != 0 you could just do a read of the position counter.

As for setting the index to rising use 
INDEXTriggerMode = 1 
when you do your config.  this tells the decoder to use rising edge of the pulse