mjs513/Teensy-4.x-Quad-Encoder-Library #5: pin0 incompatibility with multiple encoders

janbbeck, 2020-06-18:
I am using 3 of the quad encoders, and using pin0 causes the first one to malfunction.
Example:
This works:
```
//The Teensy 4.0 Encoders are supported on pins: 0, 1, 2, 3, 4, 5, 7, 30, 31 and 33
QuadEncoder encoder1(1, 1, 2, 0);  // Encoder on channel 1 of 4 available; Phase A (pin1), PhaseB(pin2), Pullups Req(0)
QuadEncoder encoder2(2, 3, 4, 0);  // Encoder on channel 2 of 4 available; Phase A (pin3), PhaseB(pin4), Pullups Req(0)
QuadEncoder encoder3(3, 5, 7, 0);  // Encoder on channel 3 of 4 available; Phase A (pin5), PhaseB(pin7), Pullups Req(0)
```

But this does not:
```
//The Teensy 4.0 Encoders are supported on pins: 0, 1, 2, 3, 4, 5, 7, 30, 31 and 33
QuadEncoder encoder1(1, 0, 1, 0);  // Encoder on channel 1 of 4 available; Phase A (pin0), PhaseB(pin1), Pullups Req(0)
QuadEncoder encoder2(2, 2, 3, 0);  // Encoder on channel 2 of 4 available; Phase A (pin2), PhaseB(pin3), Pullups Req(0)
QuadEncoder encoder3(3, 4, 5, 0);  // Encoder on channel 3 of 4 available; Phase A (pin4), PhaseB(pin5), Pullups Req(0)
```

flipping wires so that encoders are connected does nothing - encoder on channel 1 counts very erratically.
commenting out the encoder2 and encoder3 makes channel 1 with no hardware changes:
```
//The Teensy 4.0 Encoders are supported on pins: 0, 1, 2, 3, 4, 5, 7, 30, 31 and 33
QuadEncoder encoder1(1, 0, 1, 0);  // Encoder on channel 1 of 4 available; Phase A (pin0), PhaseB(pin1), Pullups Req(0)
// QuadEncoder encoder2(2, 2, 3, 0);  // Encoder on channel 2 of 4 available; Phase A (pin2), PhaseB(pin3), Pullups Req(0)
// QuadEncoder encoder3(3, 4, 5, 0);  // Encoder on channel 3 of 4 available; Phase A (pin4), PhaseB(pin5), Pullups Req(0)
```

mjs513, 2020-06-18:
Strange.  I had tested 2 encoders on  pins 0/1 and 2/3 without issue in the past.  A couple of questions:

1. Are you using the version installed with Teensyduino or did you donwload the version from Github. The Github version has the latest updates.
2. Also are you using a T4 or T4.1?

janbbeck, 2020-06-18:
I am using master from github on T4.

0/1 2/3 work, I think. it is when the third one gets put into play that problems arise. In fact, I seem to remember that 0/1 4/5 also causes the same problem.

mjs513, 2020-06-18:
Ok thanks now I understand.  Still early i will check it out during the day - have to be a setup together to test three encoders.  It may be pin config issue had to redo that got a bit complicated when i generalized it for a library.

janbbeck, 2020-06-18:
Thanks! Quick question: is there a function to set the clock rate which the QuadCounter gets? My count rates can get into many Mhz (feeding from laser interferometer). Would like to set it to the 150Mhz max.

mjs513, 2020-06-18:
Right now it uses the default IPG Clock which for the T4 at 600Mhz is 150Mhz so it should be already setup for that.  Unfortunately there is no function to change the clock for the encoders.  However with that said since the IPG clock is gated to the System Clock/4 if you overclock you can get the IPG clock up a bit.  So if you set the system clock to 828Mhz this would give you a IPG clock of 207Mhz.

janbbeck, 2020-06-18:
That is good information!
Thanks for taking the time!

mjs513, 2020-06-18:
Ok just hooked up 3 encoders and ran a simple test with pin sets 0/1, 2/3 and 4/5.  

Yep - you are right looks like i got the pin configs messed up again.  Going to take a little bit to debug and retest.  But I think I know the problem.  Its in 2 tables I have to re verify.

mjs513, 2020-06-18:
Ok figured it out, actually rather fast but figured while i was at it would retest all the pins on the T4 and T4.1

There are two pins that share a common XBAR1 IO select, pins 0 and pin 5.  So essentially they act the same.  In other words if you pin 0 you can not use pin 5 or if you use pin 5 you can not use pin 0.

Now the challenge is how to handle it - put a test in software or just make it a warning in the readme.

Hope this helps

janbbeck, 2020-06-18:
It helps a lot. Knowing this, I can comfortably ignore pin 0 and not worry about a software bug lurking. I would humbly suggest to update the readme.md for this library,
from
```
The Teensy 4.0 Encoders are supported on pins: 0, 1, 2, 3, 4, 5, 7, 30, 31 and 33.
```
to
```
The Teensy 4.0 Encoders are supported on pins: 0, 1, 2, 3, 4, 5, 7, 30, 31 and 33. Pins 0 and 5 share the same internal crossbar connections and are as such exclusive...pick one or the other.
```

I noticed you were active in a number of threads on counters on PJRC that I have been reading through. If you don't mind me picking your brain, is there a library already that allows me to feed some of the timers with external clocks and keeps a count? I need 4 64-bit counters in multi-mhz speed for the heterodyne laser interferometer (quadrature is for homodyne). So I need to either cascade timers, or use overflow interrupts to get to 64 bit. These clocks only count up, never down.

I understand if you don't have the time or inclination to answer. I have already imposed on you today and you have been very gracious, and this really isn't the right venue anyway... but I thought you have have gone through this exercise (counters with external clock) already.

In any case, thanks!

mjs513, 2020-06-18:
Wish I could answer but I don't think so, especially since you  need 4 64-bit counters in multi-mhz speed.  I would suggest that you ask on the Forum.  There are alot more knowledgeable folks on timers than me out there.