mjs513/Teensy-4.x-Quad-Encoder-Library #6: Question: use of all 4 quad encoders at the same time / pin assignments

Rob65, 2020-07-01:
I am using all 4 quad encoders at the same time.
3 quad encoders need a home signal and the 4th one an index.

The challenge is that (on a Teensy 4.1) pins 0, 5 and 37 all use XBAR1_IN17 to rout the signal to the peripheral meaning that only 11 signals can be routed at the same time.
Can this be fixed or am I hitting a hardware limit of the RT1060 chip?
I've been looking at the processor reference manual but I have not found a good overview of all possible XBAR connections

Rob

mjs513, 2020-07-01:
Actually might be able to squeeze a another pin in the library but will take me a day or 2.  Right in the middle of some home improvement projects today so tonight or tomorrow.

Rob65, 2020-07-01:
Great - just take your time, I have a number of other things to do before I finish this.

Rob65, 2020-07-02:
I might have found a solution
I still have pin 9 free (B0_11) and that can be routed to XBAR1_IN23 ALT6 with DAISY = 1

If I'm right, this needs to following additions:

`#define CORE_XIO_PIN9 IOMUXC_XBAR1_IN23_SELECT_INPUT   	//ALT6,1`

and an entry in hardware[] with the following content:
`
{???, 9, &CORE_XIO_PIN9, 6, 23, 1},`

The only thing I am not sure about is the .idx field in QuadEncoder::ENC_Hardware_t - it looks like this is not being used in the code

mjs513, 2020-07-03:
> I might have found a solution
> I still have pin 9 free (B0_11) and that can be routed to XBAR1_IN23 ALT6 with DAISY = 1
>
> If I'm right, this needs to following additions:
>
> #define CORE_XIO_PIN9 [...]

Sorry for some reason this got lost in my email.  Actually sounds like its a good solution.  the IDX is just a consecutive sequence number.  So just increment it in the two tables for the T4.0 and T4.1.

Rob65, 2020-07-04:
Mike,

sorry - I deleted that comment right after creating it. XBAR1_IN23 is actually also used for pin 30 and I confused AD_B0_11 with B0_11.
I created a list of all possible XBAR1 connections and found that there are no more pins available that can be used. So it seems I can not have all signals routed at the same time. I am now opting for a way to dynamically change pins.
The HOME signals are only used during initialization and in a sequential way (first 1 then 2 etc.)

Since no more pins seem to be available and I found an alternative connections, I think the issue can be closed.

Rob

mjs513, 2020-07-04:
Hi Rob
Not a problem been preoccupied with trying to get a 1GB NAND flash chip working on a T4.1 :)  

Glad you found a solution.  Have to check pin 32 (B0_12) its on IO10 - forgot why i didnt use it unless it does support XBARA1.

I will go ahead and close this but can always still send comments

Mike

closed 2020-07-04 by mjs513