mjs513/Teensy-4.x-Quad-Encoder-Library #10: Having some issues with getting valid data from a 3 Encoder Setup...

amaraklov, 2022-01-02:
I'm running the following code
[CODE]
// http://www.pjrc.com/teensy/td_libs_Encoder.html
//
//This example code is in the public domain.
//

#include "QuadEncoder.h"
int pullup = 1; // Weather or not to use a pullup resistor

// Change these pin numbers to the pins connected to your encoder.
// Allowable encoder pins:
// 0, 1, 2, 3, 4, 5, 7, 30, 31 and 33
// Encoder on channel 1 of 4 available
// Phase A (pin0), PhaseB(pin1), 
QuadEncoder knobA(1, 0, 1, pullup);
// Encoder on channel 2 of 4 available
//Phase A (pin2), PhaseB(pin3), Pullups Req(0)
QuadEncoder knobB(2, 2, 3, pullup);
//
QuadEncoder knobC(3, 4, 5, pullup);
//   avoid using pins with LEDs attached

void setup() {
 Serial.begin(9600);
 Serial.println("TwoKnobs Encoder Test:");
 /* Initialize Encoder/knobLeft. */
 knobA.setInitConfig();
 //Optional filter setup
 knobA.EncConfig.filterCount = 5;
 knobA.EncConfig.filterSamplePeriod = 255;
 knobA.init();
 /* Initialize Encoder/knobRight. */
 knobB.setInitConfig();
 knobB.EncConfig.filterCount = 5;
 knobB.EncConfig.filterSamplePeriod = 255;
 knobB.init();

 knobC.setInitConfig();
 knobC.EncConfig.filterCount = 5;
 knobC.EncConfig.filterSamplePeriod = 255;
 knobC.init();
}

long positionA  = -999;
long positionB = -999;
long positionC = -999;

void loop() {
 long newA, newB, newC;
 newA = knobA.read();
 newB = knobB.read();
 newC = knobC.read();
 if (newA != positionA || newB != positionB || newC != positionC) {
   Serial.print(" A = ");
   Serial.print(newA);
   Serial.print(", B = ");
   Serial.print(newB);
   Serial.print(", C = ");
   Serial.print(newC);
   Serial.println();
   positionA = newA;
   positionB = newB;
   positionC = newC;
 }
}
[/CODE]

Which is generating the following output, which I find odd, as the 3'rd channel is generating output on A and C, while the other pins are generating output on their respective channel, but only 0 and 1 or 0 and -1... while that one output is increasing/decreasing the numbers. 

A = 0, B = -1, C = 1720
 A = 0, B = 0, C = 1720
 A = 0, B = -1, C = 1720
 A = 0, B = 0, C = 1720
 A = 0, B = -1, C = 1720
 A = 0, B = 0, C = 1720
 A = 0, B = -1, C = 1720
 A = 0, B = 0, C = 1720
 A = 1, B = 0, C = 1719
 A = 1, B = 0, C = 1720
 A = 0, B = 0, C = 1720
 A = 0, B = 0, C = 1719
 A = 1, B = 0, C = 1719

mjs513, 2022-01-02:
With out know more about your set up there is one thing that I see right
off the bat.  From the ReadMe:

> WARNING! Pins 0, 5 and 37 share the same internal crossbar connections and
> are as such exclusive...pick one or the other.

What this means if you use pin 0 for one encoder you can not use pin 5 and
vice versa.  That may be causing your issue with cross talk.  Second have
you tested with just 2 encoders.  You also have not identified what
hardware you are using, IDE Version and TD version - would help with
debugging

amaraklov, 2022-01-03:
Arduino 1.8.19
Teensy Driver 1.56

amaraklov, 2022-01-03:
Ok, I also had one trace that was not working properly so all I got was 0 and -1, which was throwing me... they all work now.. thanks!

closed 2022-01-03 by amaraklov