mjs513/Teensy-4.x-Quad-Encoder-Library #11: Pins 1 and 36 on same xbar?

OliverColeman, 2022-04-11:
I think I found another crossbar limitation on pin arrangements. I'm connecting 4 AB encoders. My original intention was to use these pin assignments:
```
QuadEncoder e1(1, 1, 30)
QuadEncoder e2(2, 2, 31)
QuadEncoder e3(3, 3, 36)
QuadEncoder e4(4, 4, 37)
```
But there seemed to be weird behaviour on e1 and e3. Through trial and error I found that changing e3 to use pins 3 and 33 seemed to fix it. Looking at the diagram below it looks like pins 1 and 36 are both on xbar 16. Might need to update the readme? Happy to make a PR if my analysis is correct (and trawl through the rest of the pins to see if I can pick up any other conflicts); not that experienced so wanted to check first. 

https://github.com/KurtE/TeensyDocuments/blob/master/Teensy4.1%20Pins.pdf

OliverColeman, 2022-04-12:
Went through that diagram (after hitting what I thought was another one but turned out to be the already known one, gah). As far as I can tell there aren't any other encoder pins sharing a cross bar. Here's the pin numbers and their associated xbar, according to that diagram:

0, 5, 37 : 17
1, 36 : 16
2 : 6
3 : 7
4 : 8
7 : 15
30 : 23
31 : 22
33 : 9

mjs513, 2022-04-14:
Sorry been distracted with something else so forgot about this.  @KurtE is working on the columns now.  So probably will update the readme soon